Rahix/avr-device #5: Sed -i does not work as expected in MacOS

peacememories, 2019-05-13:
MacOS uses a different version of sed than GNU sed by default, which results in the rather cryptic error `unterminated substitute in regular expression` when running `make` in this repo. This is because MacOS sed requires a "backup extension" when using `-i`, for example `-i ''` for none. This is of course not supported by GNU sed...^^'

What should we do about this? We could instruct MacOS users to install gsed and use it in place of their normal sed, but this might break other tools that depend on the different behaviour. We could also use perl, awk or a different tool for the few replacements we have.

Rahix, 2019-05-14:
Hmm ... I'd prefer keeping sed if possible.
I don't have an OS X system to test, but apparently `sed -i'' -e '...'` (without space) should work for OS X 10.9+.  If it doesn't, `-i'.bk'` should do the trick.  Can you check whether that works?

peacememories, 2019-05-15:
`sed -i'' -e '...'` works (and i changed to code to that for testing. I'm not sure what it does to gnu-sed though.
Also it doesn't seem to do anything to fix #6

commit 7818dda6 (2019-05-16): make: Make sed usage more portable

fixes #5
closed 2019-05-16 by Rahix