Rahix/avr-device #6: unmatched visibility `pub`

peacememories, 2019-05-13:
When running `cargo +nightly-2019-05-12-x86_64-apple-darwin  doc --features atmega328p --open` i get the following error:

```
error: unmatched visibility `pub`
   --> src/devices/atmega328p/mod.rs:329:10
    |
329 | pub(crate) 0: bool = false;
    |          ^
```

Rahix, 2019-05-14:
Hmm, this might be related to #5? The only place where `pub(crate)` should appear is the line that is changed using sed in the Makefile ... I'll have a fix for that ready today in the evening, hopefully

Rahix, 2019-05-16:
The line should actually be replaced by sed to look like
```rust
pub(crate) static mut DEVICE_PERIPHERALS: bool = false;
```
From the sed expression that is used (`s/\(static mut DEVICE_PERIPHERALS\)/pub(crate) \0/`), I would guess that your sed does not replace `\0` correctly ... I am unsure why that would be the case, though

Rahix, 2019-05-16:
Can you check if commit 7818dda68e63ad511417775f3ac4bf6ae7f3c5c9 fixed this issue as well?  I changed the sed expression to use `\1` instead, in hopes that this would be more portable ...

peacememories, 2019-05-16:
It seems to work now 👍

closed 2019-05-16 by peacememories