Rahix/avr-device #8: Field Access Ranges

Rahix, 2019-05-18:
Right now, all fields which span more than one bit and which do not have a `value-group` associated in the atdf will get a range added to mark all values as allowed:

```xml
<writeConstraint>
  <range>
    <minimum>0</minimum>
     <maximum>3</maximum>
  </range>
</writeConstraint>
```

The responsible code is this:

https://github.com/Rahix/atdf2svd/blob/c7eb572c4ea12033bc2b9ccfe23d22ea2d859554/src/atdf/field.rs#L28-L40

Unfortunately, this will lead to issues further down the line, when trying to add enumerated values using `svdpatch.py`: The range will still exist, next to the values, which makes `svd2rust` believe writing this field is safe.

Right now, the only workaround is to completely remove a field and re-add it, which is error-prone and not very flexible.

Question now is, whether this is an issue here in `atdf2svd` or should rather be fixed downstream in the patch-tool.  @octycs, what is your opinion?

Rahix, 2019-05-18:
Decision was made that this will need to be fixed in the patch-tool

Rahix, 2019-05-18:
I would propose the following syntax:

```yaml
PERIPHERAL:
  REGISTER:
    _modify:
      FIELD:
        # Change the write-constraint
        # constraint is one of "enum", "none", [A, B] (range)
        _write_constraint: <constraint>
```

Rahix, 2019-05-18:
I write an implementation in c044577f27b1db3274540c1e28eb2fbb4b7dc278.  Does this look good to you, @octycs?

octycs, 2019-05-18:
This explicit syntax seems like a good solution for me.

commit 0806f412 (2019-05-18): svdpatch: Add _write_constraint modifier

"_write_constraint" allows to change the write-constraint
to one of:

- "enum": Allow only enumerated values (useEnumeratedValues)
- "none": Completely remove the <writeConstraint> tag
- [min, max]: Allow only values in a certain range (this can
  be used to make the bits() method safe)

Fixes #8
closed 2019-05-18 by Rahix