Rahix/avr-device #14: Publish on Crates.io

jonahbron, 2019-06-25:
Are there plans to publish each chip as a package to Crates.io?  Or are we waiting until the project is more stable?

Rahix, 2019-06-25:
Yes, the plan is to publish this crate (and all the generated code in it) once the basic API choices have been battle tested.  The biggest blocker right now is probably the missing interrupt handler implementation (#1).

Rahix, 2019-06-26:
Let's keep this issue open to track the effort.

Rahix, 2019-07-02:
Any other things we should definitely do before publishing?

From my side, the only missing thing is the `#[entry]` macro and a little more documentation.

jonahbron, 2019-07-02:
I think with the entry macro we're good to go.  I want to try to implement SPI this week/weekend, but that's not a blocker.  It's still useful and stable without it.

Rahix, 2020-07-26:
And it is done! :confetti_ball:

https://crates.io/crates/avr-device

closed 2020-07-26 by Rahix