Rahix/avr-device #15: Startup

Rahix, 2019-06-26:
Right now, the startup is not as pretty as I would like it to be.  Ideally, we should add a `#[entry]` procedural macro as [cortex-m](https://github.com/rust-embedded/cortex-m-rt/blob/master/macros/src/lib.rs#L85) did.

Rahix, 2019-06-28:
We should also think about pulling in the startup code written by @shepmaster in their [rust-arduino-blink-led-no-core-with-cargo](https://github.com/shepmaster/rust-arduino-blink-led-no-core-with-cargo) project instead of relying on avr-gcc.  Though this does not have a high priority for now.

jonahbron, 2019-06-28:
@Rahix What would be the material benefits?

Rahix, 2019-06-29:
Right now, not that much.  But looking into the future, it would mean independence from `avr-gcc` and `avr-libc` and ideally the ability to build the entire binary using just llvm tools.  Though I think we are still a long way away from that.

Rahix, 2019-06-30:
Oh, and now that I put some work into #19, I can tell that using custom startup code would also make the interrupt macro a lot less painful to implement.

Rahix, 2019-07-02:
I think it makes sense to add a stub `#[entry]` macro now, before publishing on crates.io, so we can later just switch the underlying implementation without breaking user code.

PR #30 (2020-07-26): Add an #[entry] macro
This is the last major thing needed for crates.io release.  Add a `#[entry]` macro which hides the actual entry implementation.  This will allow us to later transparently switch away from the avr-libc runtime to something completely custom.

The macro is similar to the one from [`cortex-m-rt`](https://github.com/rust-embedded/cortex-m-rt/blob/98e4c3c8597f1dfa52d55616c7ce7cf90682d95b/macros/src/lib.rs#L18).  However, the [automagic static mut vodoo](https://docs.rs/cortex-m-rt/0.6.12/cortex_m_rt/attr.entry.html) has not yet been added here.

Fixes #15.
closed 2020-07-26 by Rahix