Rahix/avr-device #17: Restructure Crate

Rahix, 2019-06-27:
I have noticed that my approach of mutably exclusive feature flags does not really work well.  This is most apparent from the following code:

https://github.com/Rahix/avr-device/blob/f77864dc9935b6cc402024373ec2e645d1d0cbcf/src/lib.rs#L40-L66

This code will explode with more and more chips being added.

Instead, I think we should restructure the crate to allow building for multiple chips at once.  This will never be necessary in practice but it will make building documentation (#16) and overally ergonomics much better.

My idea is to change the current structure:

```rust
mod avr_device {
    // The selected chip
    mod atmega32u4 { }
    pub use atmega32u4::*;

    pub mod interrupt { }
}
```

to the following:

```rust
mod avr_device {
    // Any chip which was selected
    pub mod atmega32u4 { }
    pub mod attiny85 { }

    pub mod interrupt { }
}
```

For downstream usage, it would change from this:

```rust
extern crate avr_device as atmega32u4;

fn main() -> ! {
    let dp = atmega32u4::Peripherals::take().unwrap();
    loop { }
}
```

to this:

```rust
pub use avr_device::atmega32u4;

fn main() -> ! {
    let dp = atmega32u4::Peripherals::take().unwrap();
    loop { }
}
```

@jonahbron, @peacememories, @octycs: Any input?

peacememories, 2019-06-27:
This feels a lot more ergonomic.

Continuing from this, would it make sense to just drop the feature flags entirely and just build for every chip? The biggest arguments against this I see are build time and emitted code size, but between caching and lto I don’t see that as too much of a prInken

Rahix, 2019-06-28:
Hmm, I don't see what we would gain from removing them entirely.  As an end-user, you will most of the time never even have to think about the flag anyway because it is set from the `at...-hal` crate.  And having only the chip you actually need available prevents you from accidentally initializing the peripherals for another microcontroller.  Or is there a reason I overlooked which you think warrants removing them?

peacememories, 2019-06-28:
I just thought it might keep things simpler. Keeping people from accidentally initializing wrong chips is a good point though.

Rahix, 2019-06-29:
Change is applied.  @peacememories, you'll have to adjust your PR as I did with the other crates in [`3b5546c77426e90409a83f6ce6974bcd4ab518dd`](https://github.com/Rahix/avr-hal/commit/3b5546c77426e90409a83f6ce6974bcd4ab518dd).

closed 2019-06-29 by Rahix