Rahix/avr-device #23: Build is not compatible with svd2rust > 0.14.0

dflemstr, 2019-08-14:
More recent versions of svd2rust emit a lot of stuff into `lib.rs` which gets overwritten by the avr-device Makefile.  I think the intent is that the output of svd2rust should be treated as a standalone crate.  Would it be possible to generate several crates and have `avr-device` depend on all of them, to avoid compatibility issues going forward?

Rahix, 2019-10-21:
Sorry for the very late response.  I attempted to fix this in commit dbbbc6efba02 ("Fix compatibility with svd2rust 1.16.1"), can you try again?

> Would it be possible to generate several crates and have avr-device depend on all of them, to avoid compatibility issues going forward?

I'd much prefer to keep the current architecture with a single crate because it keeps everything nice and compact.  The `stm32-rs` people are doing the same thing last I checked, so I think this isn't too unreasonable (They also patch the generated code ...).

dflemstr, 2019-10-24:
The approach looks reasonable to me but I have not been able to test it extensively!

closed 2019-10-24 by dflemstr