Rahix/avr-device #24: sed - expected context address on macOS

peacememories, 2019-09-21:
Hi, I've got another "this only works with gnu-sed" situation.
The issue is on this line:
https://github.com/Rahix/avr-device/blob/a2c84372a87bbfa9f71029da4fdb7268f6dc6158/Makefile#L48

`,+3c` is a gnu-sed specific command, partially explained here, i guess? https://stackoverflow.com/questions/26124650/sed-use-expected-context-address

To be quite honest I don't know what the `c` does so I'm not sure how to change this so it works on any `sed`. Maybe it's worth thinking about replacing the `sed` calls with `awk` calls altogether since `awk` seems to work the same on different posix systems.

Rahix, 2019-09-21:
Yes, this might be a good idea!  I just happened to know `sed` much better so that's what I chose ...

Another thing which should be mentioned is the fact that all this logic will need to be changed quite a bit once more because upstream `svd2rust` broke our stuff again ... Haven't had time to look at the details yet, though it seems to be quite a big change on their side.  Was reported in #23.

Rahix, 2019-09-21:
By the way, nice to see you again :)

peacememories, 2019-09-22:
I'm glad to be back :) Although I seem to have shot my rust-avr installation... again...^^ Seems to happen everytime I take a break from the project^^

Rahix, 2020-03-18:
Replaced the `sed` invocation in the above commit.  Maybe this works for you now?  Sorry for the delay ...

Rahix, 2020-06-16:
Closing for now.  If this is still a problem, we can reopen this issue!

closed 2020-06-16 by Rahix