Rahix/avr-device #25: Plan to support atmega64?

lygz5016, 2020-04-10:


Rahix, 2020-04-11:
You can add support yourself if you want to.  See commit 4f73fe2eb5a2ae66447caea74e15d45cc51645e5 ("Added atmega1280") for an example.  Things have changed a bit since then but for the most part, if you just add the new chip whereever the existing ones appear, you'll be good (e.g. search for `atmega32u4` to find all the places).

Also read ["Adding a new Chip"](https://github.com/Rahix/avr-device#adding-a-new-chip) from the README for some further infos.

If you are interested in doing this and need further infos/help, let me know!

Rahix, 2020-06-16:
I've added some very basic support for ATmega64 in commit 290613454fbd ("Add basic support for ATmega64").  I don't have any actual ATmega64 hardware on hand so I can't tell whether this works as is.  I'll have to leave testing up to you or someone else.

Rahix, 2020-10-02:
Basic support is here (untested) and no activity on this issue so I'm closing this for now.  We can reopen it later if more needs to be done.

closed 2020-10-02 by Rahix