Rahix/avr-device #28: ATxmega128A1 support

bidinzky, 2020-07-24:
Hello,

I tried to add support for the ATxmega128A1, but the ATxmega128A1.atdf I get errors while I execute make.
e.g. size not found on <register-group name="CH0" name-in-module="DMA_CH"...
Am I missing something?

Rahix, 2020-07-25:
Can you paste the full error log please?

Rahix, 2020-07-25:
Okay, from a quick glance it looks like this chip will need a few changed in [`atdf2svd`](https://github.com/Rahix/atdf2svd) ... There's some differences in the XML compared to the other files we've been working with so far.

bidinzky, 2020-07-25:
> 
> 
> Can you paste the full error log please?

This is the first error when i try with the shipped Atxmega128A1.atdf.
`SVD2RUST        ATxmega128A1
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Malformed XML. 1:1 Unexpected end of stream: no root element found', C:\Users\binde\.cargo\registry\src\github.com-1ecc6299db9ec823\svd2rust-0.17.0\src/main.rs:96:34
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
make: *** [Makefile:35: src/devices/ATxmega128A1/mod.full.rs] Error 101
`

If i format the XML i get the Error:
`ATDF2SVD        ATxmega128A1
Error: Missing attribute "caption" on
   <instance name="GPIO" ...>
make: *** [Makefile:19: svd/ATxmega128A1.bare.svd] Error 1`

Rahix, 2020-07-25:
The _missing "caption"_ error should be easy to fix.  The _size not found_ ones are a bit more difficult.  It looks like the structure in your ATDF file is different from what I've seen so far; the current code does not know how to handle that correctly.  If I understood the file correctly, the peripheral 'modules' are split into multiple XML elements which we somehow need to recombine in `atdf2svd`.  We'll have to take a closer look what exactly needs to be done ...

bidinzky, 2020-07-25:
Îs there any kind of definition for the atdf and what is needed for svd? If so i can maybe try add it myself.

Rahix, 2020-07-25:
Not that I know of :/ We worked backwards from the files and guessed how things are supposed to fit together.

If you want to take a stab at adding support for your file to `atdf2svd`, I would suggest cross-referencing [the code](https://github.com/Rahix/atdf2svd/tree/master/src/atdf) with a working file (e.g. [ATmega32U4](https://github.com/Rahix/avr-device/blob/master/vendor/atmega32u4.atdf)) and then see where the differences are in your file.  The goal is to fill the structs in [`chip.rs`](https://github.com/Rahix/atdf2svd/blob/master/src/chip.rs) with values from the atdf file.

I would suggest we open a tracking issue in `atdf2svd` and I'll also take a look and post there if I have ideas.

Rahix, 2020-07-25:
I think I found (at least one) culprit: https://github.com/Rahix/atdf2svd/issues/4

Rahix, 2020-07-25:
I've pushed some commits to `atfd2svd` which should now at least allow building _something_.  But most of that is workarounds which means that the generated SVD files is missing some register definitions and, for your case, also doesn't contain interrupt definitions.

bidinzky, 2020-07-29:
> Not that I know of :/ We worked backwards from the files and guessed how things are supposed to fit together.
>
> If you want to take a stab at adding support for your file to `atdf2svd`, I would [...]

I will look into it when i have some spare time maybe i can help.
But i guess the progress should be tracked on atdf2svd via https://github.com/Rahix/atdf2svd/issues/4.