Rahix/avr-device #49: Update release after adding new MCU

no111u3, 2020-08-17:
Could you tell me, do you have plan to create new release after add new MCU support?

Rahix, 2020-08-17:
I don't really have a release schedule, but yeah, I aim to have support for new MCUs not wait too long in the repo before issuing a new release.

no111u3, 2020-08-17:
Thanks for answer, I ask because I have some plans to update avr-hal too for support my mcu - atmega48p.

Rahix, 2020-08-17:
You can already start working on that with by patching the `avr-device` dependency in `avr-hal` to point to a locally built version.  Add this to the top-level `Cargo.toml` in `avr-hal`:

```toml
[patch.crates-io]
avr-device = { path = "/path/to/whereever/you/built/avr-device" }
```

no111u3, 2020-08-17:
I know, but for future release it needs to be update before release.

Rahix, 2020-08-20:
Version `0.2.2` is now released.

closed 2020-08-20 by Rahix