Rahix/avr-device #53: Add static-mut vodoo from cortex-m-rt

Rahix, 2020-10-02:
`cortex-m-rt` has a really nice feature where `static mut` variables in `main()` and interrupts are transformed into a safe `&'static mut _` reference.  It would be great to add the same here.

Ideally the common code can be split out from `cortex-m-rt` so we can reuse it without duplication.

couchand, 2020-10-09:
Oooh, magic!  How is that safe without an explicit `CriticalSection`?

Rahix, 2020-10-09:
It's safe because `main()` and ISRs cannot ever call themselves (to be more specific, they are not reentrant).  Thus, there will only ever be one mutable reference to the static.

couchand, 2020-10-09:
Oh I see, within a single one, not shared between them.  That makes a lot more sense.

couchand, 2020-10-09:
As a side note, if you are crazy and `sei` in your ISR it would be reentrant (or use the "avr-non-blocking-interrupt" calling convention, I'm assuming, though I haven't actually looked at that).

Rahix, 2020-10-13:
Can the same interrupt hit reentrantly with the avr-non-blocking-interrupt calling convention?  Or is it just other interrupts that could happen?

FWIW, currently we do not support this anyway but if we ever add it, we have to be very careful around that.  Thanks for mentioning!

couchand, 2020-10-13:
I can't find any authoritative source, but based on comments on AVRFreaks and reading between the lines on [the libc docs](http://www.nongnu.org/avr-libc/user-manual/group__avr__interrupts.html) it seems that yes, the nightmare scenario is possible:

> ...interrupt handlers normally do not nest. For most interrupt handlers, this is the desired behaviour, for some it is even required in order to prevent infinitely recursive interrupts (like UART interrupts, or level-triggered external interrupts).

closed 2021-06-29 by Rahix