Rahix/avr-device #60: Replace `gen-intr-lut.sh` hack with `svdtools` solution

Rahix, 2020-10-26:
Instead of the fragile sed-magic in [`gen-intr-lut.sh`](https://github.com/Rahix/avr-device/blob/master/gen-intr-lut.sh) we can use the `svd interrupts` command.  As we have [`svdtools`](https://github.com/stm32-rs/svdtools) as a dependency anyway, this would not hurt and would make the build-system much more robust.

explicite, 2020-12-02:
From `svd interrupts` we have still list only. To be generated `rs` file. Cannot see any option to move this to `rs` function.

```
jpaw@node ~/s/avr-device> svd interrupts svd/atmega1280.svd.patched 
0 RESET: External Pin,Power-on Reset,Brown-out Reset,Watchdog Reset,and JTAG AVR Reset. See Datasheet.      (in CPU)
1 INT0: External Interrupt Request 0 (in CPU)
2 INT1: External Interrupt Request 1 (in CPU)
3 INT2: External Interrupt Request 2 (in CPU)
4 INT3: External Interrupt Request 3 (in CPU)
5 INT4: External Interrupt Request 4 (in CPU)
6 INT5: External Interrupt Request 5 (in CPU)
7 INT6: External Interrupt Request 6 (in CPU)
8 INT7: External Interrupt Request 7 (in CPU)
9 PCINT0: Pin Change Interrupt Request 0 (in CPU)
10 PCINT1: Pin Change Interrupt Request 1 (in CPU)
11 PCINT2: Pin Change Interrupt Request 2 (in CPU)
12 WDT: Watchdog Time-out Interrupt (in CPU)
13 TIMER2_COMPA: Timer/Counter2 Compare Match A (in CPU)
14 TIMER2_COMPB: Timer/Counter2 Compare Match B (in CPU)
15 TIMER2_OVF: Timer/Counter2 Overflow (in CPU)
16 TIMER1_CAPT: Timer/Counter1 Capture Event (in CPU)
17 TIMER1_COMPA: Timer/Counter1 Compare Match A (in CPU)
18 TIMER1_COMPB: Timer/Counter1 Compare Match B (in CPU)
19 TIMER1_COMPC: Timer/Counter1 Compare Match C (in CPU)
20 TIMER1_OVF: Timer/Counter1 Overflow (in CPU)
21 TIMER0_COMPA: Timer/Counter0 Compare Match A (in CPU)
22 TIMER0_COMPB: Timer/Counter0 Compare Match B (in CPU)
23 TIMER0_OVF: Timer/Counter0 Overflow (in CPU)
24 SPI_STC: SPI Serial Transfer Complete (in CPU)
25 USART0_RX: USART0, Rx Complete (in CPU)
26 USART0_UDRE: USART0 Data register Empty (in CPU)
27 USART0_TX: USART0, Tx Complete (in CPU)
28 ANALOG_COMP: Analog Comparator (in CPU)
29 ADC: ADC Conversion Complete (in CPU)
30 EE_READY: EEPROM Ready (in CPU)
31 TIMER3_CAPT: Timer/Counter3 Capture Event (in CPU)
32 TIMER3_COMPA: Timer/Counter3 Compare Match A (in CPU)
33 TIMER3_COMPB: Timer/Counter3 Compare Match B (in CPU)
34 TIMER3_COMPC: Timer/Counter3 Compare Match C (in CPU)
35 TIMER3_OVF: Timer/Counter3 Overflow (in CPU)
36 USART1_RX: USART1, Rx Complete (in CPU)
37 USART1_UDRE: USART1 Data register Empty (in CPU)
38 USART1_TX: USART1, Tx Complete (in CPU)
39 TWI: 2-wire Serial Interface (in CPU)
40 SPM_READY: Store Program Memory Read (in CPU)
41 TIMER4_CAPT: Timer/Counter4 Capture Event (in CPU)
42 TIMER4_COMPA: Timer/Counter4 Compare Match A (in CPU)
43 TIMER4_COMPB: Timer/Counter4 Compare Match B (in CPU)
44 TIMER4_COMPC: Timer/Counter4 Compare Match C (in CPU)
45 TIMER4_OVF: Timer/Counter4 Overflow (in CPU)
46 TIMER5_CAPT: Timer/Counter5 Capture Event (in CPU)
47 TIMER5_COMPA: Timer/Counter5 Compare Match A (in CPU)
48 TIMER5_COMPB: Timer/Counter5 Compare Match B (in CPU)
49 TIMER5_COMPC: Timer/Counter5 Compare Match C (in CPU)
50 TIMER5_OVF: Timer/Counter5 Overflow (in CPU)
51 USART2_RX: USART2, Rx Complete (in CPU)
52 USART2_UDRE: USART2 Data register Empty (in CPU)
53 USART2_TX: USART2, Tx Complete (in CPU)
54 USART3_RX: USART3, Rx Complete (in CPU)
55 USART3_UDRE: USART3 Data register Empty (in CPU)
56 USART3_TX: USART3, Tx Complete (in CPU)
Gaps: 
```

From legacy we have full `fun` created
```
jpaw@node ~/s/avr-device> ./gen-intr-lut.sh src/devices/atmega1280/interrupt.rs
// Autogenerated.  Do not edit.
pub fn lookup_vector(chip: &str, intr: &str) -> Option<usize> {
    match chip {
        "atmega1280" => match intr {
            "RESET" => Some(0),
            "INT0" => Some(1),
            "INT1" => Some(2),
            "INT2" => Some(3),
            "INT3" => Some(4),
            "INT4" => Some(5),
            "INT5" => Some(6),
            "INT6" => Some(7),
            "INT7" => Some(8),
            "PCINT0" => Some(9),
            "PCINT1" => Some(10),
            "PCINT2" => Some(11),
            "WDT" => Some(12),
            "TIMER2_COMPA" => Some(13),
            "TIMER2_COMPB" => Some(14),
            "TIMER2_OVF" => Some(15),
            "TIMER1_CAPT" => Some(16),
            "TIMER1_COMPA" => Some(17),
            "TIMER1_COMPB" => Some(18),
            "TIMER1_COMPC" => Some(19),
            "TIMER1_OVF" => Some(20),
            "TIMER0_COMPA" => Some(21),
            "TIMER0_COMPB" => Some(22),
            "TIMER0_OVF" => Some(23),
            "SPI_STC" => Some(24),
            "USART0_RX" => Some(25),
            "USART0_UDRE" => Some(26),
            "USART0_TX" => Some(27),
            "ANALOG_COMP" => Some(28),
            "ADC" => Some(29),
            "EE_READY" => Some(30),
            "TIMER3_CAPT" => Some(31),
            "TIMER3_COMPA" => Some(32),
            "TIMER3_COMPB" => Some(33),
            "TIMER3_COMPC" => Some(34),
            "TIMER3_OVF" => Some(35),
            "USART1_RX" => Some(36),
            "USART1_UDRE" => Some(37),
            "USART1_TX" => Some(38),
            "TWI" => Some(39),
            "SPM_READY" => Some(40),
            "TIMER4_CAPT" => Some(41),
            "TIMER4_COMPA" => Some(42),
            "TIMER4_COMPB" => Some(43),
            "TIMER4_COMPC" => Some(44),
            "TIMER4_OVF" => Some(45),
            "TIMER5_CAPT" => Some(46),
            "TIMER5_COMPA" => Some(47),
            "TIMER5_COMPB" => Some(48),
            "TIMER5_COMPC" => Some(49),
            "TIMER5_OVF" => Some(50),
            "USART2_RX" => Some(51),
            "USART2_UDRE" => Some(52),
            "USART2_TX" => Some(53),
            "USART3_RX" => Some(54),
            "USART3_UDRE" => Some(55),
            "USART3_TX" => Some(56),
            _ => None,
        },
        _ => None,
    }
}
```

In old approach we are reading from pre-generated `interrupts.rs`. In new approach we should read from patched `svd` and then parse output to be compliant. Looks like  gen-intr-lut.sh  will still be there and we need to only change seed two liner with some logic.

Rahix, 2020-12-02:
The difference is that parsing the output from `svd interrupts` is much more robust than parsing the generated `interrupts.rs` file whose format can change at anytime.

Of course, we still have to generate the same `lookup_vector()` function from the `svd interrupts` output as we currently do.

closed 2020-12-04 by Rahix