Rahix/avr-device #62: ATmega4809 support

explicite, 2020-11-13:
Looks like I need to spend more time on this chip. Require probably same changes as in (#26)[https://github.com/Rahix/avr-device/issues/26#issuecomment-679831658]

@Rahix  can you please advice which topic in [atdf2svd](https://github.com/Rahix/atdf2svd) I should progress 

```
Warning: Unhandled register node: "<register-group name=\"SINGLE\" offset=\"0\" name-in-module=\"TCA_SINGLE\" union-tag-value=\"0\" ...>"
Warning: Unhandled register node: "<register-group name-in-module=\"TCA_SPLIT\" name=\"SPLIT\" offset=\"0\" union-tag-value=\"1\" ...>"
Error: Register "DIR" has a weird name!
Error: Register "DIRCLR" has a weird name!
Error: Register "DIRSET" has a weird name!
Error: Register "DIRTGL" has a weird name!
Error: Register "IN" has a weird name!
Error: Register "INTFLAGS" has a weird name!
Error: Register "OUT" has a weird name!
Error: Register "OUTCLR" has a weird name!
Error: Register "OUTSET" has a weird name!
Error: Register "OUTTGL" has a weird name!
Error: Register "PIN0CTRL" has a weird name!
Error: Register "PIN1CTRL" has a weird name!
Error: Register "PIN2CTRL" has a weird name!
Error: Register "PIN3CTRL" has a weird name!
Error: Register "PIN4CTRL" has a weird name!
Error: Register "PIN5CTRL" has a weird name!
Error: Register "PIN6CTRL" has a weird name!
Error: Register "PIN7CTRL" has a weird name!
Error: Register "PORTCTRL" has a weird name!
Error: Register "DIR" has a weird name!
Error: Register "DIRCLR" has a weird name!
Error: Register "DIRSET" has a weird name!
Error: Register "DIRTGL" has a weird name!
Error: Register "IN" has a weird name!
Error: Register "INTFLAGS" has a weird name!
Error: Register "OUT" has a weird name!
Error: Register "OUTCLR" has a weird name!
Error: Register "OUTSET" has a weird name!
Error: Register "OUTTGL" has a weird name!
Error: Register "PIN0CTRL" has a weird name!
Error: Register "PIN1CTRL" has a weird name!
Error: Register "PIN2CTRL" has a weird name!
Error: Register "PIN3CTRL" has a weird name!
Error: Register "PIN4CTRL" has a weird name!
Error: Register "PIN5CTRL" has a weird name!
Error: Register "PIN6CTRL" has a weird name!
Error: Register "PIN7CTRL" has a weird name!
Error: Register "PORTCTRL" has a weird name!
Error: Register "DIR" has a weird name!
Error: Register "DIRCLR" has a weird name!
Error: Register "DIRSET" has a weird name!
Error: Register "DIRTGL" has a weird name!
Error: Register "IN" has a weird name!
Error: Register "INTFLAGS" has a weird name!
Error: Register "OUT" has a weird name!
Error: Register "OUTCLR" has a weird name!
Error: Register "OUTSET" has a weird name!
Error: Register "OUTTGL" has a weird name!
Error: Register "PIN0CTRL" has a weird name!
Error: Register "PIN1CTRL" has a weird name!
Error: Register "PIN2CTRL" has a weird name!
Error: Register "PIN3CTRL" has a weird name!
Error: Register "PIN4CTRL" has a weird name!
Error: Register "PIN5CTRL" has a weird name!
Error: Register "PIN6CTRL" has a weird name!
Error: Register "PIN7CTRL" has a weird name!
Error: Register "PORTCTRL" has a weird name!
Error: Register "DIR" has a weird name!
Error: Register "DIRCLR" has a weird name!
Error: Register "DIRSET" has a weird name!
Error: Register "DIRTGL" has a weird name!
Error: Register "IN" has a weird name!
Error: Register "INTFLAGS" has a weird name!
[... 98 lines cut ...]
Warning: Description missing for field "VLM"
Warning: Description missing for field "CNT"
Warning: Description missing for field "PIT"
Warning: Description missing for field "CCL"
Warning: Description missing for field "LUNF"
Warning: Description missing for field "OVF"
Warning: Description missing for field "HUNF"
Warning: Description missing for field "CMP0"
Warning: Description missing for field "LCMP0"
Warning: Description missing for field "CMP1"
Warning: Description missing for field "LCMP1"
Warning: Description missing for field "CMP2"
Warning: Description missing for field "LCMP2"
Warning: Description missing for field "TWIS"
Warning: Description missing for field "TWIM"
Warning: Description missing for field "AC"
Warning: Description missing for field "RESRDY"
Warning: Description missing for field "WCOMP"
Warning: Description missing for field "EE"
Warning: Description missing for field "PORT"
Warning: Description missing for field "INT"
Warning: Description missing for field "RXC"
Warning: Description missing for field "DRE"
Warning: Description missing for field "TXC"
```

Rahix, 2020-11-13:
Hmm, Rahix/atdf2svd#6 is the main blocker I think.  I don't think it is well documented what needs to be done, though :(

explicite, 2020-11-17:
@Rahix after svd generation what I should do? How I can path svd?
Now I'm getting 
```
aw@node ~/s/avr-device> svd2rust -i svd/atmega4809.svd                          
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: In peripheral `EVSYS`

Caused by:
    0: In register `STROBE`
    1: Parsing field #0
    2: In field `STROBE0`
    3: Parsing enumerated value #0
    4: In enumerated value `EV_STROBE_CH0`
    5: Expected content in <description> tag, found none', /home/jpaw/.cargo/registry/src/github.com-1ecc6299db9ec823/svd2rust-0.17.0/src/main.rs:96:34
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Maybe I need to continue https://github.com/Rahix/avr-device/issues/26#issuecomment-679831658 ?

Rahix, 2020-11-17:
Hmm, if you remove said `<description>` tag entirely, does that help?  I think that might be another issue in `atdf2svd` ...

explicite, 2020-11-17:
Without `<description>` pass. Three files generated: `build.rs`, `device.x` and `lib.rs`.
1. Should we fix lack of description in `atdf2svd`?
2. How can I generate patched svd?
3. Should I split `lib.rs` in to different files and dirs like for other?

Rahix, 2020-11-17:
I think the right thing is option 1:  `atdf2svd` should emit correct SVD even for broken ATDF files.  This would mean not emitting empty `<description>` tags under any circumstances.

The internal representation in [`chip.rs`](https://github.com/Rahix/atdf2svd/blob/master/src/chip.rs) already allows setting `description` to `None`.  I suspect we need to add an explicit check that does this in case the ATDF contains an empty description string.

Rahix, 2020-11-17:
For the rest, take a look at this commit: 290613454fbdc5e4ac98e53deccaf74dafc88963

You need to do all these things, too.

Rahix, 2020-11-18:
I've created issue Rahix/atdf2svd#11 to track this.

explicite, 2020-11-18:
PR for  Rahix/atdf2svd#11 done
I made same changes https://github.com/Rahix/avr-device/commit/290613454fbdc5e4ac98e53deccaf74dafc88963 and pushed

Still not sure how I can create patched version of svd and then generate `.rs` files

Rahix, 2020-11-18:
> Still not sure how I can create patched version of svd and then generate .rs files

See https://github.com/Rahix/avr-device#build-instructions.

closed 2020-12-03 by Rahix