Rahix/avr-device #66: Escaping `ASYNC` register

trembel, 2020-11-26:
Trying to compile for `attiny412` following error occurs, due to the naming of the `ASYNC` register in the `TCB0` and `TCD0` peripheral:
```
error: expected identifier, found keyword `async`
   --> src/devices/attiny412/tcb0/ctrlb.rs:248:12
    |
248 |     pub fn async(&self) -> ASYNC_R {
    |            ^^^^^ expected identifier, found keyword
    |
help: you can escape reserved keywords to use them as identifiers
    |
248 |     pub fn r#async(&self) -> ASYNC_R {
    |            ^^^^^^^

error: expected identifier, found keyword `async`
   --> src/devices/attiny412/tcb0/ctrlb.rs:270:12
    |
270 |     pub fn async(&mut self) -> ASYNC_W {
    |            ^^^^^ expected identifier, found keyword
    |
help: you can escape reserved keywords to use them as identifiers
    |
270 |     pub fn r#async(&mut self) -> ASYNC_W {
    |            ^^^^^^^

error: expected identifier, found keyword `async`
   --> src/devices/attiny412/tcd0/evctrla.rs:256:12
    |
256 |     pub fn async(self) -> &'a mut W {
    |            ^^^^^ expected identifier, found keyword
    |
help: you can escape reserved keywords to use them as identifiers
    |
256 |     pub fn r#async(self) -> &'a mut W {
    |            ^^^^^^^

error: expected identifier, found keyword `async`
   --> src/devices/attiny412/tcd0/evctrlb.rs:256:12
    |
256 |     pub fn async(self) -> &'a mut W {
    |            ^^^^^ expected identifier, found keyword
    |
help: you can escape reserved keywords to use them as identifiers
    |
256 |     pub fn r#async(self) -> &'a mut W {
    |            ^^^^^^^

```

Not sure if this should be fixed in `atdf2svd` by renaming `ASYNC` registers, or here.

Rahix, 2020-11-26:
Try using a version of `svd2rust` built from git, I believe this is fixed over there.

trembel, 2020-11-26:
Yes, this is fixed - but introduces a new error (for **ALL** avr devices):
```
error[E0433]: failed to resolve: use of undeclared crate or module `cortex_m`
  --> src/devices/attiny412/interrupt.rs:52:13
   |
52 | unsafe impl cortex_m::interrupt::InterruptNumber for Interrupt {
   |             ^^^^^^^^ use of undeclared crate or module `cortex_m`

error: Compilation failed, aborting rustdoc

```

Rahix, 2020-11-26:
Hmm, that's new ... Can you try [`d6c668dedfea`](https://github.com/rust-embedded/svd2rust/commit/d6c668dedfea133ee72b05415b29a86a8f06fd10) for now?  I guess this is an upstream regression we need to report. I'll take a look ...

trembel, 2020-11-26:
Its because of this breaking change:
https://github.com/rust-embedded/svd2rust/blob/c68b71c09c87ff17270e9e8e6cb63f45276b5068/CHANGELOG.md#changed

Rahix, 2020-11-26:
Yeah, apparently https://github.com/rust-embedded/svd2rust/pull/473 was not well tested for other platforms :D

Rahix, 2020-11-26:
Makes sense, the code they modified is in the else branch of the architecture selection `if` which is also used for `AVR` ...

Rahix, 2020-11-26:
Made a report.  For the time being, using the older git hash I linked above should work for you.

trembel, 2020-11-26:
Worked, thanks.

tones111, 2021-04-13:
The fix appears to have been made upstream in https://github.com/rust-embedded/svd2rust/pull/483/files.  Since the broken change was never released, should this issue be closed?

Rahix, 2021-04-13:
Agreed.

closed 2021-04-13 by Rahix