Rahix/avr-device #68: Add support for more devices

MarkuBu, 2020-11-28:
Hi, because I want to use avr-device and avr-hal for some projects I need more supported devices. I want to add devices, beginning with "classic" ATmegas like ATmega16 and 32, ATmega2561 and some others.

Any requirements, guidelines or suggestions before I start?

Rahix, 2020-11-29:
Well, I don't have much to say ... There is a guide for adding new MCUs here: https://github.com/Rahix/avr-device#adding-a-new-chip

If you have any questions, please don't hesitate asking!

kolbma, 2020-12-10:
@Rahix I try to add support for ATmega169P. What I don't understand at the moment is when/why do I have to patch anything? I see that maybe the enum values needs a rustified name. Can you point out the other problems? Or is there already some documentation about this? Thx

Rahix, 2020-12-10:
Patching is necessary because the ATDF files often contain low-quality or plainly wrong information.  As a few examples:

- Sometimes fields are not marked as read-only/read-write when they actually are.
- Or they have a wrong bit-range in their containing register.
- Some fields expose enumerated values which are actually illegal and need to be removed.
- In some cases, the documentation is not really descriptive and needs to be updated.

Just look through the patches in `patch/common/` to get an idea.

I should note that of course we don't need to get everything right from the get-go.  Ideally, when adding initial support, look through the common patches and apply all of them which are relevant for your new MCU as well.  Anything else can be fixed later, when it comes up during HAL implementation.

Rahix, 2021-08-01:
Going to close this issue as there is no clear actionable task here.  Please open new issues (or even better: pull requests) for specific chips to be supported.

closed 2021-08-01 by Rahix