Rahix/avr-device #83: FORM step fails on `svd2rust` versions 0.18 and newer

anall, 2021-07-18:
It seems that these 2 lines are new in 0.18:

```
use generic :: * ; # [doc = "Common register and bit access and modify traits"]
pub mod generic ; # [doc (hidden)]
```

And this causes the FORM step to fail with (at signs removed to make `make` echo the commands):

```
-e 	SVD2RUST	attiny202
echo -e "\tFORM\t\tattiny202"
-e 	FORM		attiny202
RUST_LOG=WARN form -i src/devices/attiny202/mod.full.rs -o src/devices/attiny202 >/dev/null
rm src/devices/attiny202/mod.full.rs
mv src/devices/attiny202/lib.rs src/devices/attiny202/mod.rs
RUSTUP_TOOLCHAIN=nightly rustfmt src/devices/attiny202/mod.rs
Error writing files: failed to resolve mod `generic`: /home/…/avr-device/src/devices/attiny202 does not exist
make: *** [Makefile:47: src/devices/attiny202/mod.rs] Error 1
```

Rahix, 2021-07-18:
Looking at the [`svd2rust` changelog](https://github.com/rust-embedded/svd2rust/blob/master/CHANGELOG.md), I think we need to use that new `-m` switch now.

Rahix, 2021-07-19:
(In the meantime, `svd2rust` revision 56be78729279eeebef65110c13be8d96c0b9270f should work)

Rahix, 2021-08-28:
Fixed in PR #86!

closed 2021-08-28 by Rahix