Rahix/avr-device #98: docs.rs build broken for 0.3.3

Rahix, 2022-05-10:
See <https://docs.rs/crate/avr-device/0.3.3/builds/556355>.

I suspect we finally hit the memory limit.  When building locally, rustc memory usage spikes up to over 4GiB so this seems very likely...

What should we do?  I have a few ideas:

1. Only build docs with a select number of chips enabled.  I guess we could just choose the most used AVR microcontrollers and only provide online docs for those.  For other chips people can build documentation locally using
   ```rust
   cargo doc --features atmega328p --open
   ```
2. Find out why memory usage is so high?  I'm not sure if there is anything we can do about it short of changing the codegen in `svd2rust`...
3. Ask the docs.rs team to increase the limits.  Honestly, I am not a fan of this because the solution doesn't scale when more chips get added in the future...

stappersg, 2022-05-10:
All of the above. And in that order.      :smiley: 

a less lighter remark: Be aware that 4G is the limit of an `u32`.

Rahix, 2022-07-07:
Fixed with commit 1a469759a29388aa5e403cade80fb5dc2f4d8def in 0.3.4.

closed 2022-07-07 by Rahix