Rahix/avr-device #129: Arduino hangs when trying to read FUSE bits

Ayrdim, 2023-05-24:
When trying to read the fuse bits via `dp.FUSE.low.read().bits();` (it seems like `.read()` may be causing the issue) the Arduino Uno halts on that line, never continuing or panicking. 
It seems like this also happens when reading other fuse bits. 

Is this to be expected or is this a bug? I know that the fuse bits are special however I assumed that if they are accessible in then they should also be readable, or at the very least cause a panic.

Rahix, 2023-06-01:
Hi,

can you run a quick test whether other registers behave differently?  I have a feeling this might be an issue on compiler side...

Patryk27, 2023-10-11:
Depending on your uC, reading fuse bits might require extra trickery - e.g. datasheet for atmega328p says:

> 25.2.2 Reading the Fuse and Lock Bits from Softwar:
> It is possible to read both the fuse and lock bits from software. To read the lock bits, load the Z-pointer with 0x0001 and set the BLBSET and SELFPRGEN bits in SPMCSR. When an LPM instruction is executed within three CPU cycles after the BLBSET and SELFPRGEN bits are set in SPMCSR, the value of the Lock bits will be loaded in the destination register. The BLBSET and SELFPRGEN bits will auto-clear upon completion of reading the lock bits or if no LPM instruction is executed within three CPU cycles or no SPM instruction is executed within four CPU cycles. When BLBSET and SELFPRGEN are cleared, LPM will work as described in the Instruction set manual