Rahix/avr-device #138: Renesas RA4M1 support

KristijanZic, 2023-09-30:
Arduino UNO R4 WiFi featuring Renesas RA4M1 chip came out recently. Are you thinking of adding support for it?

stappersg, 2023-09-30:
On Sat, Sep 30, 2023 at 01:11:37PM -0700, Kristijan Žic wrote:
> Arduino UNO R4 WiFi featuring Renesas RA4M1 chip came out recently.

Please provide information about the board and main chip.

Groeten
Geert Stappers
-- 
Silence is hard to parse

stappersg, 2023-10-01:
> > Arduino UNO R4 WiFi featuring Renesas RA4M1 chip came out recently.

> Please provide information about the board and main chip.

https://store.arduino.cc/pages/uno-r4 says _"32-bit Arm® Cortex®-M4"_ and that is **not** an AVR-device.

> > Are you thinking of adding support for it?

Probably an impulsive question and I think it should be refined. @KristijanZic what outcome of this issue, #138, are you aiming for?

Example of a better issue:

_Arduino UNO R4 WiFi featuring Renesas RA4M1 chip came out recently. I want to help this project that it gets Rust support. Are you thinking of adding support for it?  Should I make **here** merge requests or better at **another** git repository?_

KristijanZic, 2023-10-01:
Yes

stappersg, 2023-10-01:
On Sun, Oct 01, 2023 at 01:57:45AM -0700, Kristijan Žic wrote:
> Yes

Bluf

stappersg, 2023-10-01:
> > On Sun, Oct 01, 2023 at 01:57:45AM -0700, Kristijan Žic wrote:
> > Yes
>
> Bluf

Read that as _"I don't believe that @KristijanZic wants to help here, neither does have any merge requests pending"_.  I do hope that I will be proven wrong.

KristijanZic, 2023-10-01:
> Read that as _"I don't believe that @KristijanZic wants to help here, neither does have any merge requests pending"_. I do hope that I will be proven wrong.

Get off your high horse 🤣
[image]

Rahix, 2023-10-08:
@stappersg, please dial it back a bit...

@KristijanZic, as @stappersg wrote, the RA4M1 chip is an ARM microcontroller and thus unsuitable for inclusion in the `avr-device` crate.  It needs to be supported by its own peripheral access crate.

I am not sure if anyone has already started working on Rust support for this chip. At least I haven't heard about it yet... I think you may be best off asking in the Rust Embedded community channels whether anyone already started working on this.

closed 2023-10-08 by Rahix

KristijanZic, 2023-10-09:
> @KristijanZic, as @stappersg wrote, the RA4M1 chip is an ARM microcontroller and thus unsuitable for inclusion in the `avr-device` crate.  It needs to be supported by its own peripheral access [...]

Thank you. I'm new to this so I appreciate the info. I'll ask around in the Rust Embedded or get a different board.