Rahix/avr-device #159: Error creating support for atmega16u2

zacharytomlinson, 2024-09-28:
I made the necessary changes to my knowledge following the instructions in the readme. SVD2RUST gets to the point of execution and a encounters an error for my new target:
```
-e 	GEN-VECTOR	>macros/src/vector.rs
-e 	SVD2RUST	atmega16u2
error: a value is required for '-i <FILE>' but none was supplied

```
Anyone experienced a similar issue upon integration? I suspect maybe the atdf is not in the expected format?

Rahix, 2024-09-29:
Please open a draft pull request with your current code so we can take a look.

Also, please make sure all tools are installed in the correct version as specified.

zacharytomlinson, 2024-10-16:
@Rahix from a fork? I obviously don't have a deploy key for this repo so I don't think I can open a draft pull for this one directly

zacharytomlinson, 2024-10-16:
As for dep versions:

```
~$ atdf2svd --version
atdf2svd v0.4.0 (no git)
~$ svd2rust --version
svd2rust 0.28.0 ( )
~$ form --version
form 0.12.1
~$ svd --version
svdtools, version 0.1.27
```

Rahix, 2024-10-16:
The problem was a typo in the ATDF file name which then lead to the makefile doing crazy things - see #163 and commit 5a58673 for details.

closed 2024-10-16 by Rahix