Rahix/avr-device #162: Error when adding avr-device dependency on UNO

Phirks, 2024-10-16:
When trying to enable interrupts, I ran into this error when importing avr-device:

warning: Linking globals named 'DEVICE_PERIPHERALS': symbol multiply defined!

error: failed to load bitcode of module "avr_device-e307806f1416556e.avr_device.f2dd9eb889ebcc18-cgu.0.rcgu.o":

warning: `interrupt-practice` (bin "interrupt-practice") generated 2 warnings
error: could not compile `interrupt-practice` (bin "interrupt-practice") due to 1 previous error; 2 warnings emitted

after some troubleshooting, the minimum example I can come up with is adding the following to the  sample at https://github.com/Rahix/avr-hal-template.git

cargo.toml:

`[dependencies.avr-device]
version = "0.6.0"
features = ["atmega328p","rt"]`

main.rs:

`use avr_device`

The sample then builds with no trouble, but fails when flashing using "cargo run"

downgrading avr-device to 0.5.3 fixes the issue and the code flashes and runs.

further testing shows it works as intended with the older version and different computers with fresh setups replicate the issue.

Rahix, 2024-10-16:
The problem is most likely that you have multiple versions of `avr-device` in your dependency tree when you explicitly pull in `avr-device=0.6.0`.  You can check with `cargo tree` - if multiple versions of `avr-device` show up, that's the cause of the problem.

Phirks, 2024-10-16:
I see, thank you for the help, here is my cargo tree and it looks like you're right.  I am still explicitly pulling in avr-device = 0.5.3 against the 0.5.4 in other HALs but it doesn't complain.  I'm guessing it's because it's a minor version difference vs a major version difference.  I'll continue with 0.5.4 and really appreciate all your work on these.

> PS C:\Rust\Learning\interrupt-practice> cargo tree
> interrupt-practice v0.1.0 (C:\Rust\Learning\interrupt-practice)
> ├── arduino-hal v0.1.0 (https://github.com/rahix/avr-hal?rev=3e362624547462928a219c40f9ea8e3a64f21e5f#3e362624)
> │   ├── atmega-hal v0.1.0 (https://github.com/rahix/avr-hal?rev=3e362624547462928a219c40f9ea8e3a64f21e5f#3e362624)
> │   │   ├── avr-device v0.5.4
> │   │   │   ├── avr-device-macros v0.5.4 (proc-macro)
> │   │   │   │   ├── proc-macro2 v1.0.79
> │   │   │   │   │   └── unicode-ident v1.0.13
> │   │   │   │   ├── quote v1.0.36
> │   │   │   │   │   └── proc-macro2 v1.0.79 (*)
> │   │   │   │   └── syn v1.0.109
> │   │   │   │       ├── proc-macro2 v1.0.79 (*)
> │   │   │   │       ├── quote v1.0.36 (*)
> │   │   │   │       └── unicode-ident v1.0.13
> │   │   │   ├── bare-metal v1.0.0
> │   │   │   ├── cfg-if v1.0.0
> │   │   │   └── vcell v0.1.3
> │   │   └── avr-hal-generic v0.1.0 (https://github.com/rahix/avr-hal?rev=3e362624547462928a219c40f9ea8e3a64f21e5f#3e362624)
> │   │       ├── avr-device v0.5.4 (*)
> │   │       ├── cfg-if v0.1.10
> │   │       ├── embedded-hal v0.2.7
> │   │       │   ├── nb v0.1.3
> │   │       │   │   └── nb v1.1.0
> │   │       │   └── void v1.0.2
> │   │       ├── embedded-hal v1.0.0
> │   │       ├── embedded-hal-bus v0.1.0
> │   │       │   ├── critical-section v1.1.3
> │   │       │   └── embedded-hal v1.0.0
> │   │       ├── embedded-storage v0.2.0
> │   │       ├── nb v1.1.0
> │   │       ├── paste v1.0.15 (proc-macro)
> │   │       ├── ufmt v0.2.0
> │   │       │   ├── ufmt-macros v0.3.0 (proc-macro)
> │   │       │   │   ├── proc-macro2 v1.0.79 (*)
> │   │       │   │   ├── quote v1.0.36 (*)
> │   │       │   │   └── syn v1.0.109 (*)
> │   │       │   └── ufmt-write v0.1.0
> │   │       └── unwrap-infallible v0.1.5
> │   │       [build-dependencies]
> │   │       └── rustversion v1.0.18 (proc-macro)
> │   ├── avr-device v0.5.4 (*)
> │   ├── avr-hal-generic v0.1.0 (https://github.com/rahix/avr-hal?rev=3e362624547462928a219c40f9ea8e3a64f21e5f#3e362624) (*)
> │   ├── cfg-if v1.0.0
> │   ├── embedded-hal v0.2.7 (*)
> │   ├── embedded-hal v1.0.0
> │   └── ufmt v0.2.0 (*)
> ├── avr-device v0.6.0
> │   ├── avr-device-macros v0.6.0 (proc-macro)
> │   │   ├── proc-macro2 v1.0.79 (*)
> │   │   ├── quote v1.0.36 (*)
> │   │   └── syn v1.0.109 (*)
> │   ├── bare-metal v1.0.0
> │   ├── cfg-if v1.0.0
> │   └── vcell v0.1.3
> ├── embedded-hal v1.0.0
> ├── nb v1.1.0
> ├── panic-halt v0.2.0
> └── ufmt v0.2.0 (*)
> [build-dependencies]
> └── proc-macro2 v1.0.79 (*)

closed 2024-10-16 by Phirks

Rahix, 2024-10-16:
Yeah, when specifying dependencies with `=`, cargo is allowed to choose any semver compatible release.  So specifying `0.5.3`, cargo can instead use `0.5.4`, but not `0.6.0`.  It can be very confusing at times...