Rahix/avr-device #175: Add function to get stack size

Rahix, 2025-01-06:
We should provide a function that returns the current stack size for debug analysis purposes.

(From <https://github.com/Rahix/avr-hal/discussions/589>)

LuigiPiucco, 2025-04-22:
I think this could be implemented either with inline asm (hard-coding the stack-pointer address) or by removing the atdf2svd patch which takes the SP register out from generation. To get the RAM size, I think the ATDF has it, we could patch in a constant into the respective MCU module. Or just hard-code it in the source near the `include!`, since it's just one constant.

Rahix, 2025-04-22:
I would add a function to the `asm` module which returns the current stack size.

RAM size is not something I would expose from our side - this should be derived from the linker information, usually via two symbols, one placed at start of RAM and one placed at the end.  This will give you the size that was actually reserved for the program, which may be less than the total RAM size in certain applications.

LuigiPiucco, 2025-04-23:
Sorry, by RAM size I meant having a private constant in the MCU module, so that we can do the usual `RAMEND - SP` inside the function. You're right that I could've been more clear if I'd talked about last usable RAM address instead of size, though. My mind just short-circuited to "`RAMEND = RAMSTART + RAMSIZE`, so let's call it size".

Rahix, 2025-04-23:
Yeah and the values of RAMSTART/RAMSIZE should come from the linker script is what I was trying to say, because that is the best way to find out where the actual allocated RAM region lies.

The way to do this is to define two symbols with the link names `__DATA_REGION_ORIGIN__` and `__DATA_REGION_LENGTH__` (in our case, because that's how the AVR-GCC linker scripts are set up).  You then take a pointer to these symbols and cast its address to an integer.  That gets you the origin/start of the RAM region and the size/length.

PR #189 (2025-04-30): asm: Add function to get stack size
Fix #175.

The check with `u8::MAX` is there because some MCUs (the really small ones whose data space address fits in a `u8`) may not have `SPH`. Because `extern static`s are not as optimizable as `const`s, the resulting binary can't eliminate the opposing branch, something that could be worked on.
commit 66a0b80b (2025-04-30): asm: Add function to get stack size

Fix #175
closed 2025-04-30 by Rahix