Rahix/avr-device #177: Support for ATTiny261A, ATTiny461A, ATTiny861A

mbuesch, 2025-01-12:
Tracking issue for ATTiny261A, ATTiny461A, ATTiny861A support.

Current WIP is available in this branch:
https://github.com/mbuesch/avr-device/tree/attinyx61a

Rahix, 2025-01-12:
Very nice!  Be aware of the impending changes in #157 which will affect your changeset here.  Patches will stay the same, though, now that we have moved to rust `svdtools` (which you have already pulled into your WIP branch I see)

mbuesch, 2025-01-12:
Thanks for the hint. I will rebase on #157 once that is merged.
I need a bit of time to polish this anyway. But it seems to basically work as-is.

mbuesch, 2025-01-12:
One question regarding the svd toolchain:
The devices have a couple of fields that are split across two registers (ADMUX and ADREFS). Does the svd toolchain have support for that?
I currently chose to not implement the upper bits of these fields. Which is fine for now, IMO.

Rahix, 2025-01-12:
No, split fields cannot be modelled, unfortunately.  So far, we have dealt with this by giving the bigger part "double names".  Although in practice that HAL code for such chips doesn't reference any enum names at all and instead just writes bare integers.  See here for example:

https://github.com/Rahix/avr-hal/blob/d02ab21d17f06baadb5c0f2abf7cf3ea9c0a35b3/mcu/atmega-hal/src/adc.rs#L289-L322

PR #179 (2025-06-05): Add support for ATtiny261A, ATtiny461A, ATtiny861A
This PR adds support for the ATTiny x61A family devices.

I have tested this on an ATtiny861A device.
The 2 and 4 variants are extremely similar with less RAM and Flash.

These devices share the same physical footprint with the ATTiny26. Therefore, the port declaration is shared.

Closes: #177
commit 7c569a73 (2025-06-05): Add support for ATtiny261A, ATtiny461A, ATtiny861A

Closes: #177
closed 2025-06-05 by Rahix