Rahix/avr-device #187: Build script should only parse ATDF for selected MCUs

Rahix, 2025-04-18:
@LuigiPiucco, so I just noticed one more thing that slipped through review ;)

In `find()`, you already parse **all** ATDFs (and also all patches).  This happens before filtering for selected MCUs which means that a lot of unnecessary parsing work is done for MCUs that won't actually get code-generated.

I noticed this through suspiciously long build-times for single-MCU builds and also while playing with the error messages (injecting an error into the ATDF of an MCU that is not selected still leads to to a build error).

Do you want to take care of this?

LuigiPiucco, 2025-04-18:
I'll need some time to work on personal things right now, so no, but I'll be back in a day or two.

Rahix, 2025-04-18:
No worries, I did not mean for this to be urgent at all (after all, you know my response times....).  Just wanted to know whether you are interested in picking this up at some point when you have time, or if you would rather see someone else put in the effort :)

PR #188 (2025-04-21): build: only parse selected MCU inputs
Fix #187.

Since I had some extra time, I went ahead and took care of this. It's a simple change, I merged `InputFinder::filter_from_features` and `InputFinder::find`. We still iterate every path, to add the name to the list of available MCUs in case of error, but the inputs are only inserted if the variable is present.
closed 2025-04-21 by Rahix